OrcaSlicer/OrcaSlicer #301: Easy way to migrate from Superslicer, import .ini settings/config bundle from superslicer to Bambu Studio

trohnjavolta, 2023-02-10:
**Is your feature request related to a problem? Please describe.**
Not being able to easily import my configs from superslicer into Bambu Studio is a big showstopper for me, maybe others feel the same.

**Describe the solution you'd like**
An easy option to import settings/configs from Superslicer. Right now Superslicer exports settings/configs as .ini file while Bambu Studio only takes .json files.

**Describe alternatives you've considered**
Already tried to convert .ini files in .json but importing them via Bambu Studio import option doesn't work and if I manually place them in the config folder, after a restart, Bambu Studio seems to delete those files.

hliebscher, 2023-02-10:
superslicer is to different... softfever has many new features with settings are not in superslicer and superslicer has many other features .... 
you can easy modify the default settings

trohnjavolta, 2023-02-10:
> superslicer is to different... softfever has many new features with settings are not in superslicer and superslicer has many other features .... you can easy modify the default settings

I understand, still even some half assed way that pops up errors for the parameters that couldn't be imported would imo be so much better than setting up everything all over again.

hliebscher, 2023-02-11:
The effort to produce all the messages and catch errors is too great, and then the next bug report comes in asking why the super slicer profile doesn't work the same way it does in super slicer.

The best way would be to create a printer from the super slicer profile using a script, such as Python, which can then easily be integrated into the software.

closed 2023-11-25 by github-actions[bot]