OrcaSlicer/OrcaSlicer #302: Add Custom GCODE to layers

rfnovo, 2023-02-11:
**Describe the bug**
When adding custom GCODE to a layer, it will not show that there is a GCODE there, you must hover mouse over for it to highlight the area. It will also not allow to view the GCODE, now allow you to edit it. 

**3mf File for This Bug**
Any file

**To Reproduce**
Steps to reproduce the behavior:
1. Slice any STL
2. In Preview tab, on the side bar, go to any layer, right click and select "Add Custom G-code"
3. Move slide bar to other layer. 
4. The layer that you add the GCODE has no indication not mark. 

**Expected behavior**
It should display the layer with the GCODE, and allow you to edit it. Bonus points if also shows in the sliced part.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 10
 - 1.4.4

hliebscher, 2023-02-11:
On Mac OS, you need to go to the "Prepare" tab and then back to "Preview" in order to see the change. It could also be a problem with the graphics card driver in Windows, which is often the case.

rfnovo, 2023-02-12:
Tried that with no improvement. Might be an issue with card driver, but Prusa and SS can show that normally.

SoftFever, 2023-02-18:
Looks like a rendering glitch caused by the graphic driver.
[image]

vicpon, 2023-05-11:
I have the same problem on Windows. I can see the little bars that show where the custom gcode is but I can't edit/view the existing gcode.

[image]

SoftFever, 2023-05-11:
@vicpon can you share the project?

vicpon, 2023-05-11:
> @vicpon can you share the project?

[ringing_tower.zip](https://github.com/SoftFever/OrcaSlicer/files/11452485/ringing_tower.zip)

closed 2023-11-25 by github-actions[bot]

skl111, 2023-12-12:
Is there a nightly build where this has been addressed?