OrcaSlicer/OrcaSlicer #304: ** BUILD FAILED **  macos

hliebscher, 2023-02-11:
The following build commands failed:
        CompileC /Users/hliebscher/github/BambuStudio-SoftFever-1/build/build/libslic3r_gui.build/Release/Objects-normal/arm64/Plater.o /Users/hliebscher/github/BambuStudio-SoftFever-1/src/slic3r/GUI/Plater.cpp normal arm64 c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'libslic3r_gui' from project 'BambuStudio')

after the last 2 patches the compiler has errors 

MAC M1 13.1

hliebscher, 2023-02-14:
@SoftFever  slic3r/GUI/Plater.cpp:8132:17: error: no viable overloaded '='
        *max_lh = { layer_height };

this is the error

SoftFever, 2023-02-15:
Okay, let me boot up my Mac machine:)Sent from my iPhone

hliebscher, 2023-02-15:
i have fix this in #314 PR

closed 2023-02-16 by hliebscher