OrcaSlicer/OrcaSlicer #306: Version 1.4.4 often stopped working in Windows 11

jeor112, 2023-02-12:
This program version often stopped working. Didn't see this issue in the earlier versions.
It often happens after slicing and after having send the job to the printer.

[image]

 - OS: Windows 11
 - Version SoftFever V. 1.4.4

hliebscher, 2023-02-12:
witch printer ? have you problems with other programs or internet connection?

jeor112, 2023-02-12:
It's X1C and I haven't noticed internet connection problems with other devices. If I'm using the original BambuStudio I haven't that issue.

hliebscher, 2023-02-12:
do you use a special feature ? can you send the 3mf as zip file?

jeor112, 2023-02-12:
No, I'm using standard profiles for the the X1C and as fare as I know without any special features. The program shuts down with several 3mf and stl files loaded.
Another issue I have experienced is that after the latest firmware update, I cannot use the Flow calibration. First layer is coming out very badly when I do. When calibration is off, I got no prolems. Off cause, I don't think it's related to the slicer. But strange things happens.
I think I would try to uninstall the SoftFever program and reinstall from scratch. I will report the outcome of this.

hliebscher, 2023-02-13:
the best is clear the softfever config directories and check GPU driver if not running GPU Tweak !

jeor112, 2023-02-15:
Same issue after clearing the config directories and reinstalling the software.
What do you mean about GPU tweak?
I have a NVIDIA Geforce RTX 3060 Ti. What would you suggest that I should tweak?

Thanks

jeor112, 2023-02-15:
After clearing/deleting the config (C:\Users\jeor\AppData\Roaming\BambuStudio-SoftFever) and re-installed the SoftFever version (C:\Program Files\BambuStudio-SoftFever_V1.4.4) and I still got the same problems, I then switch over to use the origanal Bambu Studio version 1.4.2.13 and haven't got any problems at all.

With the SoftFever version I also have seen, that It has problems to detect and edit the spools in the AMS.
There is also problems when switching on the camera. It takes very long time to show up.

Yet another issue I have seen with the SoftFever is the Flow calibration.
In SoftFever the print turns out very badly, but do I skip the calibration, the print is much better.
In the original version, there is no problems when using the flow calibration.

I hope it makes sense to you.
I don't understand if I'm the only one with this kind of problems.
Have been using the SoftFever for some time and my problems only started with the 1.4.4 version and the latest firmware.

I'm not an expert, but I have the feeling, that it all started after the latest firmware update (X1-C: 01.04.00.00, AMS: 00.00.05.96)

hliebscher, 2023-02-17:
sorry for that but i have no bambu printer and use only mac and my snapmaker IDEX works perfectly and waste no filament during color change 
But BambuLab has many changes in last time for the ams ... pls test the 1.4.3 again against the problems ... sometimes helps a firmware update 
sorry that i can't help

closed 2023-11-25 by github-actions[bot]