OrcaSlicer/OrcaSlicer #308: OSX - Check for Updates

pacmac, 2023-02-13:
Latest version of BBS, lates version of OSX.

The check for updates feature under the help menu appears to do nothing or is not working.

After selecting this, nothing is reported, but it is on the original BBS version.

hliebscher, 2023-02-13:
if is no new softfever version than nothing to download but the softfever version not replace the BBS version !
two different apps with different config directories

pacmac, 2023-02-13:
I mean there is no popup message saying whether there is or is not an update available, nothing at all happens.

On the official version, a message pops up saying whether or not an update is available.

hliebscher, 2023-02-14:
the update check is completly different like org BambuStudio is like Superslicer we can check if a new version 1.4.5 is ready

closed 2023-11-25 by github-actions[bot]