OrcaSlicer/OrcaSlicer #315: Can´t open any BambuStudio for Mac

SA7BNT, 2023-02-15:
Hey,
just tested different releases for Mac_x86_64. But i can´t get none of them to open. 
I´m always getting 
[image]

I´m running 
[image]

Did i miss something to install??

EuanWebster, 2023-02-15:
Same issue for me Macbook Pro 16" 2020, macOS Ventura x86.

Munza2020, 2023-02-15:
Same as first post. Macbook Pro 15" 2017 Ventura 13.1

EuanWebster, 2023-02-15:
> Hey, just tested different releases for Mac_x86_64. But i can´t get none of them to open. I´m always getting [image]
> 
> I´m running [image]
> 
> Did i miss something to install??

Are you able to raise this to a bug?

hliebscher, 2023-02-15:
Have you read the note on the Main page about this issue and the articel that are linked? https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/

same at https://github.com/SoftFever/BambuStudio-SoftFever/issues/274

EuanWebster, 2023-02-15:
> Have you read the note on the Main page about this issue and the articel that are linked? https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/
> 
> same at #274

Thank you, that worked perfectly!!

SA7BNT, 2023-02-15:
> Have you read the note on the Main page about this issue and the articel that are linked? https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/
> 
> same at #274

Well i didn't get it so much attention to that post because i just read M1. Mine is not a M1. So maybe change the note to all Mac user´s 
Thanks for the hint. Was working for me to.

closed 2023-02-15 by SA7BNT

Munza2020, 2023-02-16:
Thanks, that worked, but yes, I ignored note because it said for Apple M1 users as well. It is just for Apple users, the chip type is irrelevant.