OrcaSlicer/OrcaSlicer #318: Artillery Genius

mw8010, 2023-02-16:
Hello,
is it possible to add the artillery genius to the database?

Thanks very much.
MW

hliebscher, 2023-02-16:
witch printer has you now in software selected? 
and use 
M190 S[bed_temperature_initial_layer_single]
M109 S[nozzle_temperature_initial_layer]
G28
not absolut values

hliebscher, 2023-02-16:
[BUG] V86 echo message triggered during print #24281
it is a bug in Marlin witch version do you use?

mw8010, 2023-02-16:
At the moment Marlin 2.1.1 is running on the printer.  
I chose the Artillery X2 as a base and adjusted the settings to match the Genius.

hliebscher, 2023-02-16:
pls export the g-code from cura and bambu and compare the start section or it is a bug from marlin eeprom crc

mw8010, 2023-02-16:
I've now spent the last few hours creating a printer profile for my Genius in the Bambu Slicer (Bambu original).
Was able to do a successful test here too. So I don't quite understand why the fork shouldn't work.
The only problem I had with my test file was that Bambu constantly used the g17 code when changing the layer and I didn't get behind it to turn it off.
I also looked at the compiled gcode files between Softfever and Bambu and saw differences here too. I have no idea if something is wrong with the slicing here. Sorry, I'm not a programmer :-)

What I can still try is to port the profile that I built for the Bambu to the Softfever and try it again here.

lg MW

hliebscher, 2023-02-17:
pls send the 3mf from bambu and softfever as zip attach this it is only a config problem in the profile not in start code i think  in layer change code...

mw8010, 2023-02-17:
Good morning,
I am enclosing the relevant files in ZIP format.
[Test_Cube.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10764876/Test_Cube.zip)

hliebscher, 2023-02-19:
@mw8010 try this   or we make a call in GERMAN ;-)
 

;START CODE
M83						; extruder relative mode
G28 				        ; home all axes
M109 S[first_layer_temperature]                                           ; hotend temperature
M140 S[first_layer_bed_temperature] 	                                ; heatbed temperature
M190 S[first_layer_bed_temperature] 	                                ; wait for the bed to heat up
M109 S[first_layer_temperature]                                            ; wait for the extruder to heat up
G92 E0 					                ; reset extruder
G1 X20 Y5 Z0.3 F5000.0		       ; move to start-line position
G1 Z0.3 F1000 				       ; print height
G1 X200 Y5 F1500.0 E15 	                ; draw 1st line
G1 X200 Y5.3 Z0.3 F5000.0			; move to side a little
G1 X5.3  Y5.3 Z0.3 F1500.0 E30 		; draw 2nd line
G1 Z3 F3000 			            		; move z up little to prevent scratching of surface

:END GCODE
G91 			;Relative positionning
G1 E-2 Z0.2 F2400 		;Retract and raise Z
G1 X5 Y5 F3000 		;Wipe out
G1 Z10 			;Raise Z more
G90 			;Absolute positionning
G1 X0 Y100 		;Present print
M106 S0 			;Turn-off fan
M104 S0 			;Turn-off hotend
M140 S0 			;Turn-off bed
M84 X Y E 			;Disable all steppers but Z

closed 2023-11-25 by github-actions[bot]