OrcaSlicer/OrcaSlicer #321: Unable to run under macOS 12.6

macminoz, 2023-02-16:
hi there, I downloaded twice but I got the same error:
[image]

even with terminal no luck

`macmino@ MacOS % strings ./BambuStudio
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun`

hliebscher, 2023-02-16:
Have you read the note on the Main page about this issue and the articel that are linked? https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/

same at https://github.com/SoftFever/BambuStudio-SoftFever/issues/274

macminoz, 2023-02-17:
this solved my issue! thanks

closed 2023-02-17 by macminoz