OrcaSlicer/OrcaSlicer #322: unable to print directly from slicer with pull request #319

davidja92, 2023-02-17:
**Describe the bug**
when using pull request #319 nightly, the print slices fine but when pressing the print plate or print all button nothing happens. I can send the file to the printer wirelessly then print from the screen. i understand its a nightly just wanted to point it out thanks

**3mf File for This Bug**
any file

**To Reproduce**
Steps to reproduce the behavior:
use nightly pull request #319 and try to print directly from the slicer.

**Expected behavior**
pressing print plate or print all sends the .3mf to the printer and auto starts the print

**Screenshots**

**Desktop (please complete the following information):**
 - OS: windows 11 
 - Version 1.4.4 nightly #319

hliebscher, 2023-02-17:
do you have merge the #319? but the newly added calibration tests, ringing,vfa and temp tower are in [7fe5b84](https://github.com/SoftFever/BambuStudio-SoftFever/commit/7fe5b84c41746a9fd2a73c09644f83f975283453)
[last week](https://github.com/SoftFever/BambuStudio-SoftFever/commit/7fe5b84c41746a9fd2a73c09644f83f975283453)
not in 319
witch printer do you use?

SoftFever, 2023-02-17:
Thanks for the feedback.
Fixed.

SoftFever, 2023-02-18:
Closing as it is fixed.
Feel free to reopen it in case you still have issues

closed 2023-02-18 by SoftFever