OrcaSlicer/OrcaSlicer #324: 3rd Party IP Camera Support via RTSP

km4lkx, 2023-02-17:
SoftFever Dev Team, 

I have a Bambu Lab P1P and the camera module is straight garbage with no ability to upgrade to the better camera from BBL. Several members of the community have designed 3rd party camera mounts to mount wifi cameras to. A lot of these cameras support RTSP streams. 

I assume that RTSP is currently being used to monitor a printer as that would be the easiest way to monitor a camera in the application. However, I haven't really looked at the code to see how it is being handled. If it is, then all that would need to be added is the ability to select a stream via IP and port via a menu button above the camera box in Device>Status.

This would allow P1P owners as well as anyone running Klipper on low resource hardware such as an RPi Zero the option of using higher resolution and higher framerate cameras.

MasterTom99, 2023-02-21:
+1
As many other P1P user did, I added a camera to the printer that is independent of the printer itself. I used a ESP32-CAM. Right now I watch the videostream using VLC on my laptop or my iPhone. It would be much easier to get the stream shown in Bambu-Studio directly.

jbeardon, 2023-02-22:
+1 as well
There are also plenty of cheap IPCams (eg TPLink Tapo C110) that natively expose an RTSP stream.
This would be most useful.

pedrodonkey, 2023-03-14:
+1 here too. 

Support for viewing a RaspberryPi cam running off of a pi zero inside Bambu Studio would be awesome.

ab3dgeek, 2023-03-14:
yes please !

QuittyMR, 2023-03-26:
+1

As a point for consideration; this, plus a custom layer-change hook (custom RTSP request or even custom code) would completely open this to 3rd party cameras, enabling both monitoring and timelapse shots.

beeinaboats, 2023-03-29:
+1 here too,

been trying to find a slicer that would let me use rstp for a third party camera in the gui

RadsWa, 2023-05-12:
oh yes please!

closed 2023-11-25 by github-actions[bot]

sneakyjoeru, 2024-08-01:
I also want rtsp

Klaster1, 2024-10-29:
RTSP support would be nice indeed.

sneakyjoeru, 2024-10-29:
@SoftFever can we please have this back to requests?

Adaaks, 2024-11-09:
There is now a way to add a tapo RTSP camera to orca, check this out: https://www.reddit.com/r/BambuLab/comments/1gnl9s8/guide_link_tapo_camera_to_orca_slicer/

Klaster1, 2024-11-10:
@Adaaks I'm basically doing the same with Node.js, but would definitely prefer a built-in feature.