OrcaSlicer/OrcaSlicer #330: Webcam traffic issue/bug Mac OS version?

SA7BNT, 2023-02-19:
Hey,

first of all thanks for this great slicer program.
I´m running Mac OS and version 1.4.4 of the BambuStudio-SoftFever version on a Mac.
Inside the slicer im using the "Device" panel to remote my printer.
It seems that the webcam stream has a problem. Every time i reopen a different webcam (2 webcams active for that printer) Bambu adds the new open webcam stream to the old one. 
So if you switch 3,4,5 times all traffic will be added and you will get at least about 10MB/s (single stream of webcam about 2MB/s) 
Also if im not active in the Device menu/panel but lets say in the Prepare menu/panel i still can monitor traffic so Bambu will not kill the stream if I switch in the Bambu menu.
I notified this because I´m monitoring my network. 
I also verified this traffic problem with the std. Mainsail installation. I don't get this problem there. So it seems to be a Bambu issue.

If I´m wrong please let me know 

Best 
Tim

hliebscher, 2023-02-19:
witch printer ? and witch panel ?  MAC M1 or x64 witch os version ?

SA7BNT, 2023-02-19:
> witch printer ? and witch panel ? MAC M1 or x64 witch os version ?

Printer is an Prusa MK3s and Mac is x86_64 Big Sur Version 11.7.4 (20G1120)
[image]

hliebscher, 2023-02-19:
the os version is very old ... at 13.1 it works for me

SA7BNT, 2023-02-19:
> the os version is very old ... at 13.1 it works for me

Sorry, but explain why it will not run with this version of os. What will be the different from Mainsail standalone (there it works with the traffic) to Bambu which seems to use a http connection to Mainsail (Activity Monitor is showing that in the BambuStudio Networking Process under "Open Files and Ports).  For me it looks like there is a problem in the Bambu engine. 

If you are one of the developer it will be an easy way to explain for me, if not lets wait for an answer from one of the developer.
Hope we will get an answer.

hliebscher, 2023-02-19:
i am not the developer sorry i have many bugs fixt ... i wanted to help ...

SA7BNT, 2023-02-21:
Now running on macOS Ventura 13.2.1 and still get that problem. changed.

closed 2023-11-25 by github-actions[bot]

SA7BNT, 2023-11-25:
Still having the same issue with Ventura 13.6.1 x86_64