OrcaSlicer/OrcaSlicer #331: Voron origin center of bed

BlackStump, 2023-02-20:
I have tried on a couple occasions to use this slicer but not being able to set the Origin to the bed center is hold me up.
I have setup the Voron 300x300 to have the Origin in the center of the bed same as the Delta printer I have.
[image]
looks like it is just about there - X origin is greyed out while Y origin I can change to 150

hopefully a simple fix

thanks

rovili, 2023-02-20:
You might set up something wrong.

The origin (0.0) of a Voron, in any size, must be at the left lower corner. You cannot set it up the same way as a Delta, because the Delta must have the origin in the middle of the bed.

BlackStump, 2023-02-20:
This is the only slicer known to me that you can't set the Origin to the bed center.
Slicers that can be set to bed center 
PrusaSlicer
SuperSlicer
Ideamaker
Icesl
Kisslicer
Simplify3D

BlackStump, 2023-02-20:
examples
[image]
[image]
[image]

SoftFever, 2023-02-22:
It needs some work to support changing the origin(not the UI). I intentionally cripple the UI so UI won't accidentally change it.

BlackStump, 2023-02-23:
fair enough

closed 2023-02-23 by BlackStump