OrcaSlicer/OrcaSlicer #332: Wierd slow down on 3 identical objects

Prutsium, 2023-02-20:
[Latest 1.4.5 beta]
I noticed a weird slowdown on 3 identical objects where no slowdown would be required.
Sliced with normal perimeter generator so not Arachne.

Of course i can switch back to classical mode slowdown handling but it's wierd that this happens.
[image]

When sliced on classic mode slowdown all look good:
[image]

3mf attached (zipped)
[slowdown.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10781377/slowdown.zip)

zviratko, 2023-02-20:
I was just about to file the same thing
I noticed slowdown where they were not before (and IMO not necessary but that comes up to the settings - assuming it's caused by the new "slow down for overhangs" implementation)
However doing it for only one of the objects is weird. In my case it looks like this:
[image]

I think it
1) should be less aggressive out of the box if it's wanted behaviour
2) should be consistent :-)

zviratko, 2023-02-20:
Disabling Arachne but using the new slow down generation looks like this - looks more correct but still inconsistent with the same object:
[image]

rovili, 2023-02-20:
I have the exact same issue. Duplicated objects are slowed down to the minimal speed in the filament settings.

SoftFever, 2023-02-20:
Yeah, it's a bug that only happened on cloned objects.
I have just fixed it [c868798](https://github.com/SoftFever/BambuStudio-SoftFever/commit/c8687989e5eecab43fdfeee4a2b7cb52f87a31e7)

SoftFever, 2023-02-20:
This is the bug that delayed 1.4.5 release :(

Prutsium, 2023-02-20:
> This is the bug that delayed 1.4.5 release :(

This contains the fix? https://github.com/SoftFever/BambuStudio-SoftFever/actions/runs/4225839174

Prutsium, 2023-02-20:
Ah just noticed it contains [c868798](https://github.com/SoftFever/BambuStudio-SoftFever/commit/c8687989e5eecab43fdfeee4a2b7cb52f87a31e7)

psiberfunk, 2023-02-21:
Lol, also just noticed this.  Seems to only happen with classic slowdown algorithm  Here's 3 project files for replication, showing it happens with arachne or classic wall generator, but turns off when you turn off the new overhang speed algorithm:

[NewSlowDownCodeBug_ClassicWalls_ClassicOverhangSlowdownAlgorithm.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10789048/NewSlowDownCodeBug_ClassicWalls_ClassicOverhangSlowdownAlgorithm.zip)

Prutsium, 2023-02-21:
Looks good with the latest Nightly!
[image]

SoftFever, 2023-02-22:
Looks like we have sorted out the problem.
Closing it.

closed 2023-02-22 by SoftFever