OrcaSlicer/OrcaSlicer #334: Gap infill is too aggressive

rovili, 2023-02-20:
The Gap infill is very aggressive and fills spots where actually nothing is to fill and it produces "zits".

It would be great if you could give us, such as SuperSlicer does, the possibility to adjust the Gap Infill threshold.

Please see an example with SoftFever:
[image]

With SuperSlicer:
[image]

[image]

hliebscher, 2023-02-20:
can you zip the 3mf an attach for testing?

SoftFever, 2023-02-21:
We have gap fill filter feature.
[image]
However, this option won't affect top/bottom layers at the moment.
1.4.5 will allow filtering out of all gap fills.

rovili, 2023-02-21:
That is great news! Thanks a lot for your effort in this project! =)

SoftFever, 2023-02-22:
Looks like we have sorted out the problem.
Closing it.

closed 2023-02-22 by SoftFever