OrcaSlicer/OrcaSlicer #343: Temperature Tower Not Manifold

jbeardon, 2023-02-22:
**Describe the bug**
Slicer reports that the temperature_tower.stl  has 18 non-manifold edges

It prints OK but hardly ideal!

[image]

SoftFever, 2023-02-22:
It's a bit tricky as the non-manifold edges are created when splitting the model.
This doesn't cause any issue though.

jbeardon, 2023-02-22:
Ah, fair enough. Could the test not be produced using scripted height modifiers, removing the requirement to split the model?
Just a thought :-)

hliebscher, 2023-02-25:
SF fixed in the last patch ... for MAC ist works THANKS

SoftFever, 2023-02-26:
Fixed in [463393c](https://github.com/SoftFever/BambuStudio-SoftFever/commit/463393c96952d67f7dd8aa8a0d476f6e36092ba5)

We can arrange and stack each block in the code too. 
But I found an easy way out ;)

closed 2023-02-26 by SoftFever

sdylewski, 2024-01-09:
Still broken in 1.9.0 on OSX

SoCuul, 2024-05-01:
Still broken in 2.0.0 on macOS

EYEhaveYOU95, 2024-10-02:
There is still the same ERROR: 14 non-manifold edges.

Currently in Orca Slicer 2.2.0-beta 

I hope it still works as anticipated with using REPAIR, as repair itself works.

shauneccles, 2025-01-04:
@SoftFever this is still an issue in 2.2.0 release - can you reopen or shall I log another issue?

SoCuul, 2025-01-06:
> @SoftFever this is still an issue in 2.2.0 release - can you reopen or shall I log another issue?

You should probably re-open. Would bring it to the top of the issues list as well.

3dRikal, 2025-03-09:
Still an issue 2.2.0 Win

GAZ082, 2025-03-29:
Issue in Linux with 2.2.3

marauderlabs, 2026-03-04:
Version 2.3.1 this issue persists. (I noticed at least on the portable one)