OrcaSlicer/OrcaSlicer #344: MAC Versions both ARM and x86 are corrupt.

Busterdick, 2023-02-22:
Downloaded both Versions to try on my MAC. Both Versions x86 and ARM do crash with the message that the programme itself is damaged.

Tried on my x86 Core i9 MAC 27" and on my Macbook Pro M1Max ARM.

No Error message except:

BambuStudio-SoftFever" is damaged and cannot be opened. It is recommended to move the object to the recycle bin

Any ideas?
 

*Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

suchabadperson, 2023-02-22:
In the description of each release it is explained how to make it work on macOS.
https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/

SoftFever, 2023-02-22:
For now please use the command `xattr -d com.apple.quarantine /path/to/app.app` to allow running unsigned app on Mac.
I'm investigating how to sign it. Hopefully, we will have a signed app for the next release.

Naninani, 2023-02-22:
> For now please use the command `xattr -d com.apple.quarantine /path/to/app.app` to allow running unsigned app on Mac. I'm investigating how to sign it. Hopefully, we will have a signed app for the next release.

@SoftFever, Its actually a 2 way process. 1st is the code signing, where you could use the xtools xcrun codesign command with a valid cert.
2nd is notarization - for that you'd need a valid Apple Developer Account.
Notarization is very important as it sends the bits to the apple notarization service that checks code is correctly signed, and doesn’t contain malicious content.
For both to work, you'd need to generate the .plist files having the required info for each file that needs to be signed and notarized. If you miss a file, the signing will succeed, but the notarization will fail.

suchabadperson, 2023-02-22:
About the Apple version, it would be nice to take a look at the size of the icon that appears in the Launchpad, which is larger than the rest.

[image]

CrazyCreator13, 2023-02-24:
i can, with the xattr command, start the app and working. but all changes go away after i close the app and reopen :(

hliebscher, 2023-02-25:
@CrazyCreator13   you open the app at application directory?

CrazyCreator13, 2023-02-26:
Yes, of course.
It was currently working. Can it be that BambuStudio-SoftFever saves the settings in the cloud? Because when I'm logged into the app, I have my settings. At least that's how I feel right now.

SoftFever, 2023-03-15:
> 

I've finished the code sign part but run into trouble(error 65) when I try notarizing it.
It may be because of the incorrect .plist as you mentioned.
Is there a magic command/script to automatically populate .plist by any chance? ;)

closed 2023-11-25 by github-actions[bot]