OrcaSlicer/OrcaSlicer #345: Top surface pattern not updated on 1.4.5

suchabadperson, 2023-02-22:
According to the changelog from the latest version, the top surface pattern was changed to monotonic by default, but it is still as monotonic line. I have tested on both macOS and Windows to be sure.

hliebscher, 2023-02-22:
for me it works MAC M1 OS 13.1...

SoftFever, 2023-02-24:
> According to the changelog from the latest version, the top surface pattern was changed to monotonic by default, but it is still as monotonic line. I have tested on both macOS and Windows to be sure.

Oops, I forgot to mention I only made this change for third party printers.
The monotonic line pattern generates aggressive gap fills which may cause trouble for slightly slow printers

closed 2023-11-25 by github-actions[bot]