OrcaSlicer/OrcaSlicer #349: Extra perimeter every second layer

DorjeDorf, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
This is a feature that is present in a couple slicers like Kisslicer and in SuperSlicer. 
It allows you to have alternating amount of perimeters between layers. 
In Kisslicer you would type in 3.5 loops and it would add an extra loop every second layer. 
In SuperSlicer, there was a checkbox that enabled it. This causes the infil to run further over the extra perimeter and add strength to the part. Would love to see it here. 

**Describe the solution you'd like**
The easiest thing would be to move the code from SuperSlicer - I'd assume it can be done. It's in Quality, Extra perimeters, On odd layers. 

**Describe alternatives you've considered**
Using another slicer, but that's not quite the goal here ;) 

**Additional context**
Nothing more to add. Would love to see this here.

psiberfunk, 2023-02-24:
Cura has this too IIRC

DorjeDorf, 2023-02-24:
Yes, sure, it's a choice between better external layers or better part strength. But I would imagine that some people (like me) would be willing to sacrifice the outer walls a bit (I am fairly sure I haven't seen it being THAT pronounced) for stronger parts.

Misterke, 2023-08-11:
The biggest benefit to me is how it allows "sandwiching" every other layer of sparse infill at the sides of the infill with those extra layers.  Especially if you have a modifying volume that causes the top and bottom layers to be 0 at some places and hence has the sparse infill become visible and structural, then you want that infill to be well attached to the rest of the object and sandwiching every other layer between such extra walls can do just that.

stewe12, 2023-09-06:
I would also love this feature. It is really significant change in part rigidity and much stronger bond between infill and outer walls. And after flow and PA calibration I had no visible artefacts on outer walls. But I am usually using 3 or more walls.

Skivaks, 2023-09-21:
please add this. very usefull

oleksii-suprun, 2023-10-29:
This is the only feature left that prevents me from entirely moving from Cura to OrcaSlicer. I like that feature very much in Cura because it helps me to cover the tiny gap between the sparse infill and the inner wall when I need 100% infill. That "sandwich" will help to cover the seam between the infill and the wall in every other layer and this will reduce the chance that the wall will be broken off from the infill.

oleksii-suprun, 2023-10-29:
Here is a link to the correspondent SuperSlicer issue: https://github.com/supermerill/SuperSlicer/issues/89. 
And commit: https://github.com/supermerill/SuperSlicer/commit/004ec5480d38ec7fb3a456e70fb63e0c3892b9d6

The change does not look complicated. I could even try to port it to OrcaSlicer by myself, but, unfortunately, I'm not a C++ developer and I'm not sure if I can make it with good quality :)

ChaosBlades, 2023-11-23:
Been waiting for this to be added. IIRC the last thing I am missing from SuperSlicer. Just had a part rip out the infill perfectly on me. Could have really used these extra layers to lock it in. Re-printing with 100% infill to be sure this time.

"this telegraphs through the outer walls." Don't see much point in using this with low numbers of walls. This setting is for super strong parts with like 4+ walls then this adds an extra 5th wall to lock in a high 40%+ infill for super strong parts without resorting to 100% infill. Not for something you are using 2 walls with and 15% infill.

DorjeDorf, 2023-12-25:
Added here: https://github.com/SoftFever/OrcaSlicer/pull/3196

closed 2023-12-25 by DorjeDorf