OrcaSlicer/OrcaSlicer #351: FR: Extrusion width as percentage

OXERY, 2023-02-23:
Hey,

at first: Thanks for your great work, really amazing :-)
In superslicer, I was mainly using percentages of the Nozzle width as extrusion width, mainly because [Ellis Print Guide](https://ellis3dp.com/Print-Tuning-Guide/) was recommending it and it was great to not have to deal with many settings when changing from e.g. a 0.4 to a 0.8 nozzle.
If you could add this, this would be a great addition and would save me a few profiles to add.

Thanks :-)

SoftFever, 2023-02-23:
Yeah, this is already on my todo list for the next release as I like percentages for line width too :)

ar1a, 2023-05-16:
Really keen for this! I think this feature is the one thing that's keeping me from switching from SS

EvaFPV, 2023-05-20:
Pleaseeeeeeeee implement this as percentage of NOZZLE WIDTH like SS does and not percentage of layer height like PS, so one profile can be used without adjustments for various nozzle sizes. Have no clue what the point of PS's implementation is lol.

jeremytodd1, 2023-06-08:
I 100% agree with @EvaFPV.

Basing the percentages off of the nozzle width, like how SS handles it, is a much better way to go than basing it off of the layer height, like PS.

SS makes it so simple to make profiles for different nozzle sizes. PS it is a pain to do.

This feature is one of my most looked forward to features for OrcaSlicer.

closed 2023-08-13 by SoftFever