OrcaSlicer/OrcaSlicer #356: Acceleration values not calculated in speed visualization

hutje, 2023-02-23:
**Describe the bug**
When changing the acceleration values, the visualization for the color scheme are not being used in the calculation for the speed.
The slicing is correct and the time is right, but not in the speed visualization.

**3mf File for This Bug**
[Final Top twist container simulation speed not correct.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10818985/Final.Top.twist.container.simulation.speed.not.correct.zip)

[image]
[image]
[image]

**Desktop (please complete the following information):**
Windows 10 - BambuStudio SoftFever (v1.4.5)

psiberfunk, 2023-02-24:
I’m not sure this is a bug per se , it’s just a feature (modeling acceleration behavior of the printer ) that doesn’t exist in the preview ?

SoftFever, 2023-02-24:
The visualization is for speed only per design.

hutje, 2023-02-24:
When changing the max volumetric speed, the visualization changes accordingly... Seems to me acceleration would also be important to give a good visualization of the speed you are actually get during printing...

psiberfunk, 2023-02-24:
I think what he’s asking for @SoftFever is that the speed display model the expected effects of acceleration on speed such that you’d see the expected speed gradient on a given layer, as it would appear in reality.

presently there is an assumption that the acceleration is instantaneous in the speed display .

psiberfunk, 2023-02-24:
To clarify , this is clearly a new feature and the display would have different troubleshooting purposes .  If you implemented this , You would probably want to add a totally new visualization type like : “Speed, with accelerations”

psiberfunk, 2023-02-24:
> When changing the max volumetric speed, the visualization changes accordingly... Seems to me acceleration would also be important to give a good visualization of the speed you actually get during printing...

This is because the GCODE for feed-rate actually changes in this case.  What you’re asking for is to visualize not just the GCODE setting in this case but ALSO to model the impact the currently set acceleration and jerk would have on the line drawn.  It’s a much more complex thing because you have to make additional look backs to find the present acceleration and additional modeling of speed over visualization segments , and then decide at what resolution you should model those things .

hutje, 2023-02-24:
Thanks for the explanation. Didn't think about that.

closed 2023-11-25 by github-actions[bot]