OrcaSlicer/OrcaSlicer #358: SoftFever icon is too large on Mac OS

moebis, 2023-02-24:
Screenshot shows it all, but basically the SoftFever version of the logo/icon doesn't conform to the standard Mac OS size for these icons, it's too large: [https://imgur.com/a/lw1BLOl](https://imgur.com/a/lw1BLOl)

Installed latest 1.4.5 version too.

moebis, 2023-02-24:
It also shows large in the launcher once you close the app too: [https://imgur.com/a/wrqUJmg](https://imgur.com/a/wrqUJmg)

hliebscher, 2023-02-24:
in the launcher only during application start

Snicket8247, 2023-02-24:
I made a version with the right dimensions if you want to use that.
I also gave it a gradient instead of a solid color since I find it more aesthetically pleasing.

[image]

moebis, 2023-02-25:
> I made a version with the right dimensions if you want to use that until it is fixed. I also gave it a gradient instead of a solid color since I find it more aesthetically pleasing.
> 
> You can right click the application and press "get info", then simply select the icon in the top right and paste it in.
> 
> [image]

Thank you. Looks great. FYI, once you launch it the icon goes back to the non-gradient, but at least it stays proportional to the other icons. In a way it's kinda cool, when it's not running it's darker (gradient) and launched it's the lighter solid color. Neat unintentional effect.

hliebscher, 2023-02-25:
thanks for the png file ;-)

closed 2023-03-17 by SoftFever