OrcaSlicer/OrcaSlicer #360: X-Y shrink compensation

tapir, 2023-02-24:
**Is your feature request related to a problem? Please describe.**
Printing ABS parts comes up 0.6% smaller than intended design due to shrinkage

**Describe the solution you'd like**
We need a new filament setting where one can put a shrinkage percentage for X-Y and the every model imported that uses the said filament should be automatically resized on X-Y axis by that amount. We can ignore the Z shrinkage (as do many other slicers) as it's very very little.

**Describe alternatives you've considered**
Resizing objects manually or designing the parts oversized. Both are cumbersome and does not transfer well if you want your parts to be printed by somebody else too.

**Additional context**
Most slicers already have it. Some of them have it in print settings which is not ideal as shrinkage is strictly a material property thus it would be best to have it under filament settings.

boellerbenny, 2023-02-25:
Hello, I see it the same way tapir describes it. I would be incredibly excited to have this feature.

kostja89, 2023-02-25:
that would also convince me to switch :)

SoftFever, 2023-03-17:
Implemented in 1.5.0

closed 2023-03-17 by SoftFever