OrcaSlicer/OrcaSlicer #363: Font unreadable in Save preset

TexZeTech, 2023-02-24:
**Version**
1.4.5

**Description**
The font is a dark color and is very hard to read in the Save preset window.

**To Reproduce**
1. Go to Prepare tab
2. Click on 'G gear' icon and click 'Edit Preset'
3. Click on 'Save current Machine'
4. Text input font color will be a black on gray background.

**Expected behavior**
To match the font color of the current theme in the application.

**Screenshot**
[image]

**Desktop & hardware**
Operating System: Manjaro Linux
KDE Plasma Version: 5.26.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 5.15.94-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

TexZeTech, 2023-03-18:
As of OrcaSlicer V1.50 It is still affected by this bug, It also affects the g-code text boxes as well as the post-processing scripts text windows.

TexZeTech, 2023-04-07:
This seems to be fixed in 1.6.1 

I tested this in a fresh VM running Manjaro.

closed 2023-04-07 by TexZeTech

shyblower, 2023-04-30:
I'm on 1.6.2 and the problem still persists!