OrcaSlicer/OrcaSlicer #367: **LINUX** App crashes when video stream play button is pressed and other warning popup

tinymachines3d, 2023-02-24:
Printer connects without issue but crashes when you want to view the video stream (bambu x1c)

**To Reproduce**
Steps to reproduce the behavior:
1. download appimage
2. install with AppImageLauncher
3. login to Bambu account
4. Device> Play button
5. Install gstreamer full suite and retry with same results

**Expected behavior**
video stream should open as in Bambu Slicer app

**Screenshots**
[image]
[image]

~$ hostnamectl
Operating System: Pop!_OS 22.04 LTS               
Kernel: Linux 6.1.11-76060111-generic
Architecture: x86-64
Hardware Vendor: MSI
Hardware Model: MS-7A68

~$ lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 05)
00:01.0 PCI bridge: Intel Corporation 6th-10th Gen Core Processor PCIe Controller (x16) (rev 05)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller
00:14.2 Signal processing controller: Intel Corporation 200 Series PCH Thermal Subsystem
00:16.0 Communication controller: Intel Corporation 200 Series PCH CSME HECI #1
00:17.0 SATA controller: Intel Corporation 200 Series PCH SATA controller [AHCI mode]
00:1b.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #17 (rev f0)
00:1b.4 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #21 (rev f0)
00:1c.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #1 (rev f0)
00:1c.4 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #5 (rev f0)
00:1c.7 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #8 (rev f0)
00:1d.0 PCI bridge: Intel Corporation 200 Series PCH PCI Express Root Port #9 (rev f0)
00:1f.0 ISA bridge: Intel Corporation 200 Series PCH LPC Controller (Z270)
00:1f.2 Memory controller: Intel Corporation 200 Series/Z370 Chipset Family Power Management Controller
00:1f.3 Audio device: Intel Corporation 200 Series PCH HD Audio
00:1f.4 SMBus: Intel Corporation 200 Series/Z370 Chipset Family SMBus Controller
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (2) I219-V
01:00.0 VGA compatible controller: NVIDIA Corporation GM107GL [Quadro K2200] (rev a2)
01:00.1 Audio device: NVIDIA Corporation GM107 High Definition Audio Controller [GeForce 940MX] (rev a1)
03:00.0 Non-Volatile memory controller: Phison Electronics Corporation E12 NVMe Controller (rev 01)
05:00.0 USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller
06:00.0 Network controller: Qualcomm Atheros AR93xx Wireless Network Adapter (rev 01)
07:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980

sudermanjr, 2023-03-12:
I am also seeing the exact same issue on a just-installed ubuntu 22.04 system.

```
 uname -a
Linux desktop 5.19.0-35-generic #36~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 17 15:17:25 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

sudermanjr, 2023-03-12:
I just checked and the upstream bambu studio has this issue as well, even on the latest release.

tinymachines3d, 2023-03-12:
I tried on another system (amd64) and 1.50.59, 1.50.60 and SoftFever 1.4.5 all work properly. The system I had issues on was intel based. So, it may be intel related but I'm going to double check the gstreamer packages on the intel machine when back in office. Below are the results when I send commands

`gst-inspect-1.0 --gst-version`

GStreamer Core Library version 1.20.3

`dpkg -l | grep gstreamer`

ii  gir1.2-gstreamer-1.0:amd64              1.20.3-0ubuntu1                                                 amd64        GObject introspection data for the GStreamer library
ii  gstreamer1.0-alsa:amd64                 1.20.1-1                                                        amd64        GStreamer plugin for ALSA
ii  gstreamer1.0-clutter-3.0:amd64          3.0.27-2ubuntu1                                                 amd64        Clutter PLugin for GStreamer 1.0
ii  gstreamer1.0-gl:amd64                   1.20.1-1                                                        amd64        GStreamer plugins for GL
ii  gstreamer1.0-gtk3:amd64                 1.20.3-0ubuntu1                                                 amd64        GStreamer plugin for GTK+3
ii  gstreamer1.0-libav:amd64                1.20.3-0ubuntu1                                                 amd64        ffmpeg plugin for GStreamer
ii  gstreamer1.0-pipewire:amd64             0.3.66~1676588331~22.04~135fd49                                 amd64        GStreamer 1.0 plugin for the PipeWire multimedia server
ii  gstreamer1.0-plugins-bad:amd64          1.20.3-0ubuntu1                                                 amd64        GStreamer plugins from the "bad" set
ii  gstreamer1.0-plugins-base:amd64         1.20.1-1                                                        amd64        GStreamer plugins from the "base" set
ii  gstreamer1.0-plugins-base-apps          1.20.1-1                                                        amd64        GStreamer helper programs from the "base" set
ii  gstreamer1.0-plugins-good:amd64         1.20.3-0ubuntu1                                                 amd64        GStreamer plugins from the "good" set
ii  gstreamer1.0-plugins-ugly:amd64         1.20.1-1                                                        amd64        GStreamer plugins from the "ugly" set
ii  gstreamer1.0-pulseaudio:amd64           1.20.3-0ubuntu1                                                 amd64        GStreamer plugin for PulseAudio (transitional package)
ii  gstreamer1.0-tools                      1.20.3-0ubuntu1                                                 amd64        Tools for use with GStreamer
ii  gstreamer1.0-vaapi:amd64                1.20.1-1ubuntu1                                                 amd64        VA-API plugins for GStreamer
ii  gstreamer1.0-x:amd64                    1.20.1-1                                                        amd64        GStreamer plugins for X11 and Pango
ii  libgstreamer-gl1.0-0:amd64              1.20.1-1                                                        amd64        GStreamer GL libraries
ii  libgstreamer-plugins-bad1.0-0:amd64     1.20.3-0ubuntu1                                                 amd64        GStreamer libraries from the "bad" set
ii  libgstreamer-plugins-base1.0-0:amd64    1.20.1-1                                                        amd64        GStreamer libraries from the "base" set
ii  libgstreamer-plugins-good1.0-0:amd64    1.20.3-0ubuntu1                                                 amd64        GStreamer development files for libraries from the "good" set
ii  libgstreamer1.0-0:amd64                 1.20.3-0ubuntu1                                                 amd64        Core GStreamer libraries and elements

sudermanjr, 2023-06-18:
Just wanted to confirm that I am still seeing this on v1.6.3 of orca slicer. Logs aren't super helpful, `Illegal instruction (core dumped)` is all I see.

closed 2023-11-25 by github-actions[bot]