OrcaSlicer/OrcaSlicer #368: Temperature Tower non-manifold and Fix Model not available in Linux

WZ9V, 2023-02-25:
**Describe the bug**
The generated temperature tower is non-manifold (Error 18 non-manifold edges) and the Fix Model feature in Linux is not available.   

**To Reproduce**
Calibration -> Temp Tower
Slice

**Expected behavior**
Either a manifold temp tower should be generated and/or the Fix Model feature should be made working.

**Desktop (please complete the following information):**
 - OS: Linux Mint 21.1 Cinnamon 5.6.7, Kernel  5.15.0-60-generic
 - SoftFever 1.4.5

SoftFever, 2023-02-26:
duplicated with #343

closed 2023-02-26 by SoftFever

MermaidMel, 2024-08-15:
I found a solution was to download an older version as I use flash forge Ocra this was available https://industry.flashforge.com/download-center/148 worked