OrcaSlicer/OrcaSlicer #370: Slicer setup wizard blank

DVSVIDEO, 2023-02-25:
**Describe the bug**
The slicer not start

**3mf File for This Bug**
Not start, not 3mf
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'BambuStudio-SoftFever_V1.4.5_Win64'
2. Click on 'Bambu-studio.exe'
3. start software, appear the setup wizard screen, and nothing to see
4. This blank screen not allow next step

**Expected behavior**
some choice config an else

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win10 enterprise 64bit

SoftFever, 2023-02-25:
Can you install these two runtimes:
https://github.com/SoftFever/BambuStudio-SoftFever#2-if-you-have-troubles-to-run-the-build-you-might-need-to-install-following-runtimes

DVSVIDEO, 2023-02-26:
Ohhh .... thanks!
the edge runtime solve the problem:
[image]

closed 2023-02-26 by DVSVIDEO