OrcaSlicer/OrcaSlicer #371: Enable / disable dark mode

suchabadperson, 2023-02-25:
I would like to propose the possibility of including a box in the preferences menu to be able **to enable or disable dark mode** without depending on the OS.

Ibot-11, 2023-02-25:
There is already a box. As far as i know this is already independent from the os.
[image]

suchabadperson, 2023-02-25:
That's true! It seems that macOS version lacks that feature. I wonder if it would be possible to include it.

closed 2023-11-25 by github-actions[bot]

jimdoescode, 2024-09-03:
Sorry to resurrect this but there's still no dark mode option in OSX. Are there any plans to add this?

reevo1985, 2025-03-26:
Would like to also add there is no dark mode on Linux so this is a Windows only feature unfortunately.