OrcaSlicer/OrcaSlicer #372: Wipe Inside at Start/End of perimeter (from SuperSlicer)

pemerick07, 2023-02-25:
**Is your feature request related to a problem? Please describe.**
When printing with external perimeters first, a blob is formed at the start of the perimeter. This is even more of a problem when trying to get 100% dense parts for fluid-tightness because more excess nozzle pressure typically exists at deretract.

**Describe the solution you'd like**
SuperSlicer has a feature that allows for the nozzle to start inside the external perimeter by a wipe depth (nozzle % or absolute) where the deretract is performed. Then the nozzle moves towards the actual starting point of the perimeter to start printing. With a bit of tuning of the wipe depth, the blob is almost eliminated. 

**Describe alternatives you've considered**
I've tried inner-outer-inner-infill, but when there aren't more than 2 perimeters, the outside perimeter is printed first and forms a blob.

**Additional context**
[image]

pemerick07, 2023-03-18:
@SoftFever If you tell me where to look in the code I can probably implement and test this feature.

cinusik, 2023-04-21:
This is the one thing that keeping me from fully switching to Orca from SS. There is a big difference in seams visibility in favor to SS with that wipe inside functionality.
So, +1 for that. It will be great to have it in OrcaSlicer 👍

pemerick07, 2023-11-22:
@SoftFever I love using Orcaslicer, but I'm still really missing this feature. If you can implement it, I'll test it and buy you lots more coffee :)

SoftFever, 2023-11-22:
I will try to port this feature in 1.9

pemerick07, 2023-12-21:
> This one - so this will eliminate all “possible” blobs? Any sideeffects when enabled? I

I don't know if I'd say all possible blobs, but most. In my experience when using SuperSlicer with this option and aligned seams I was able to get very smooth seams without blobs even when over-extruding. I print many fluid-tight parts which requires some over-extrusion, so printing external perimeters first is essential to keep the subsequent over-extruded lines contained, but it results in messy seams using Bambu Slicer and OrcaSlicer on my X1C. One of the benefits of using the wipe inside method over the inner-outer-inner method is that it works with only two perimeters.

igiannakas, 2023-12-26:
@SoftFever  if you haven't made a start at this one I'll try to port it over tomorrow.

Naninani, 2023-12-26:
> I don't know if I'd say all possible blobs, but most. In my experience when using SuperSlicer with this option and aligned seams I was able to get very smooth seams without blobs even when [...]

Sam here - my application consists of 2 walls only. I do polymer clay cutters, where the main wall consists of 2 perimeters and the cutting edge is just one. There are always some blobs on the seams, no matter the settings.

> @SoftFever if you haven't made a start at this one I'll try to port it over tomorrow.

This feature will be a game changer for me. Thx for looking into implementing it!

igiannakas, 2023-12-26:
> Sam here - my application consists of 2 walls only. I do polymer clay cutters, where the main wall consists of 2 perimeters and the cutting edge is just one. There are always some blobs on the [...]

Draft PR is open: https://github.com/SoftFever/OrcaSlicer/pull/3287

Can test it from the nightly builds. It's very much work in progress but the basics work. De-retract inside the model by half a nozzle width and start the external perimeter immediately thereafter.

SoftFever, 2023-12-26:
> @SoftFever  if you haven't made a start at this one I'll try to port it over tomorrow.

No, I haven't started this. 
I'm still ghosted with the sudden Mac x86 crash issue...

closed 2023-12-29 by pemerick07