OrcaSlicer/OrcaSlicer #377: FEATURE REQUEST: Show BOTH Imperial and Metric measurements inside the scale box

Photogad, 2023-02-25:
**Is your feature request related to a problem? Please describe.**
I would like the software to show BOTH metric AND imperial measurements inside the scale box. This is very important for us USA customers who receive measurements from our clients in imperial (inches) measurement, but as 3D printer professional we work with metric. This will make it easy for us to convert!!! Without having to google the conversion every time we need to convert.

**Describe the solution you'd like**
Add extra boxes in the scale window for metric/imperial so both are visible. OR you could add a little "checkbox" there (LIKE PRUSASLICER HAS!!!!) to switch between imperial/metric without having to go all the way to File -> Preferences -> Measurement Units -> OK   every time!!! Very slow that way! And not convenient!

**Describe alternatives you've considered**
The only other alternative is google to convert imperial to metric every time but this is very slow and annoying.

**Additional context**
[image]

Photogad, 2023-05-14:
bump
lol

EklDuderino, 2023-09-03:
I 2nd this addition. Prusa slicer has a check box that toggles on the fly between metric and imperial measurements. This would be a great improvement. Please add this feature. Thank you!

closed 2023-12-10 by github-actions[bot]