OrcaSlicer/OrcaSlicer #379: Only one wall on topmost layer with arachne enabled

liftbag, 2023-02-26:
**Describe the bug**
In the latests dev builds, when Arachne wall generator is enabled, the option "Only one wall on top surfaces" produces results only on the topmost layer.

**Expected behavior**
As with the Classic wall generator, each exposed top layer should have a single perimeter.

**Screenshots**

Classic wall generator

[image]

Arachne wall generator

[image]

**Desktop (please complete the following information):**
 - OS: Apple macOS 
 - Version Ventura 13.2.1
 - Bambu Studio SoftFever nightly build 374 build mac x64 93

liftbag, 2023-02-26:
I just realized the same issue is present in SuperSlicer. To enable the single perimeter on the top surfaces with arachne in SuperSlicer it must be enabled from the classic mode, as with arachne enabled the option is currently greyed out. When you re-enable arachne the option works, but only on the topmost layer.

hliebscher, 2023-02-26:
@liftbag pls look at #350

liftbag, 2023-02-26:
> @liftbag pls look at #350

Yes, in the letest dev releases the issue #350 is solved while the single perimeter on top layers is only partially solved.

Ibot-11, 2023-02-26:
Single wall top surface will be included in the next release
#340

SoftFever, 2023-02-27:
can you attach a 3mf project file to repro the issue?

kvnper, 2023-03-01:
This issue also occurs with the setting "Only one wall on **first** layer"

kvnper, 2023-03-01:
I am additionally having an issue where no top layers at all were getting a single wall.

liftbag, 2023-03-01:
> I am additionally having an issue where no top layers at all are getting a single wall.

I've changed the title, according to the fact that the issue refers to the latest dev builds. The single perimeter on the bottom and topmost layer with arachne enabled currently works fine on the latest dev build. But the rest of the top layers under the maximum Z height, set the perimeters established in the settings.

kvnper, 2023-03-01:
> 

ok no problem, although in my opinion i suggest putting it at the start of the title instead of the end

liftbag, 2023-03-07:
Solved in the build Mac x64 no. 102

[image]

Thanks a lot! Can be closed.

the0bone, 2023-05-15:
> I've changed the title, according to the fact that the issue refers to the latest dev builds. 

But it is not just the latest dev Build.
v 1.6.2 have the issue as well.

liftbag, 2023-05-15:
> But it is not just the latest dev Build. v 1.6.2 have the issue as well.

The post is dated February 26th. I believe it was a development version of 1.6.2.
The issue refers to the fact that Arachne Wall generator broke the single wall on both the top and bottom layers. A fix was later made which is the current way it works.

liftbag, 2023-07-06:
It seems that PrusaSlicer is finally going to implement this feature in the next release. It would be nice if OrcaSlicer fixes it before then.

https://github.com/prusa3d/PrusaSlicer/issues/10913#issuecomment-1609484522

SoftFever, 2023-07-07:
> It seems that PrusaSlicer is finally going to implement this feature in the next release. It would be nice if OrcaSlicer fixes it before then.
> 
> [prusa3d/PrusaSlicer#10913 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/10913#issuecomment-1609484522)

I assume you meant "port it after then" ;)
Thanks for the heads up. I'm glad to hear that.
I've been a bit busy with family and work recently, so I have to deprioritize some feature requests.

liftbag, 2023-08-30:
Seems fixed with the latest 1.7 dev release :)

[image]

Jo-Con-El, 2023-12-05:
> Seems fixed with the latest 1.7 dev release :)
> 
@liftbag: Could you please close this issue then?  Thanks!

closed 2023-12-06 by liftbag