OrcaSlicer/OrcaSlicer #381: BUG: Some windows buttons are missing text (white text on white button)

Photogad, 2023-02-26:
**Describe the bug**
Some windows buttons are missing text (white text on white button)

**To Reproduce**
Check the "object type" (modifier) menu when using dark mode

**Expected behavior**
The button text is visible and dark mode not white text on white button

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 13
 - Bambu Slicer SoftFever 1.4.5

rufinus, 2023-03-22:
OS: Ubuntu (Dark mode)
Slicer: OrcaSlicer 1.5.0

- Highlighted/Active Tab - Text color equals background color
- Textareas - Text color almost background color

see screenshot
[image]

guestisp, 2023-04-16:
Same in 1.6.1 for linux

Photogad, 2023-05-21:
@SoftFever I am not sure if this is fixed in newest 1.6.3 nightly release (I am still on 1.6.2) but as below you can see the buttons still have no text in 1.6.2
[image]

closed 2023-11-25 by github-actions[bot]