OrcaSlicer/OrcaSlicer #386: Add terminal/console

rkarlsba, 2023-02-27:
I'm used to using OctoPrint and I really like the way I can 'talk' gcode with the printer using the terminal (or console if you like). This way, I can test things out before hardcoding everything. Could this be done?

kylek29, 2023-02-28:
That's in line with something I was thinking a bit about. BLabs appears to want to maintain as much of their proprietary nature as possible, so they're reluctant to introduce any core features that would threaten it (API access, Plugin ability, etc.). SoftFever has the same connection logic to the printer, so theoretically, an "Octoprint" like bridge might be possible. 

If a stdout pipe can be established so that secondary programs can get status and interact with the printer (just like it would if connected in the software -> Device)? This might open an avenue for creating a bridge to the printer for other developers to work upon, especially since SoftFever works on Linux (iirc), that bridge might be a Printer --> <Local Wifi> --> <RPi w/a SoftFever API version> --> <Third-Parties can handle the rest>

This would also allow your console interface, because an embedded console could just pull in the stdout.

cramsvik, 2023-09-22:
Depending on your printer/firmware (I suppose) you can have this by adding the device in Orca (available at least in 1.7.0). it will give you full access to the webpage for the printer, including the console. I run klipper on my Voron, and have the same interface inside OrcaSlicer as I do in my webbrowser. No more looking for the right browser window and tab; it is right there inside Orca! :-D

closed 2023-12-30 by github-actions[bot]