OrcaSlicer/OrcaSlicer #387: random crashes to desktop

IrishMoss, 2023-02-27:
**Describe the bug**
I'm getting random crashes to desktop.  As far as i can tell, it only happens when printing, so i'm assuming it's something to do with the monitoring data transmission.  I've tried to replicate it on 'vanilla' bambu studio, but have not had any crashes there.

Is there any way to enable any kind of advanced logging or a debug mode, to try and capture what it's doing at the time of the crash?

**3mf File for This Bug**
N/A

**To Reproduce**
Random occurrences makes it difficult to reproduce.  Usually takes a while before it happens, and the only app correlation I've found is it happens when a print job is running.

**Expected behavior**
obvious.

**Screenshots**
N/A

**Printer model**
P1P

**Desktop (please complete the following information):**
Windows 10 (build 19044.2604)

IrishMoss, 2023-02-28:
Ok, so i was able to find out where the windows crashdumps are, downloaded windbg, and had it analyze one of the crash dumps.  
Looks like some kind of access violation error?

dump and analyzed output attached.
thanks.
[studio_exception_log.txt](https://github.com/SoftFever/BambuStudio-SoftFever/files/10852895/studio_exception_log.txt)
[bambu-studio.exe.18280.dmp.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10852989/bambu-studio.exe.18280.dmp.zip)

julie777, 2023-06-02:
I get crashes multiple times a day. If I can get the build working I am going to start debugging them.

shayded-exe, 2023-06-04:
I too am getting random crashes with nothing in the log to show for it.

Half the time it won't even recover my backup when it starts up again and I'll lose work!!

limited660, 2023-06-09:
I'm on Windows 11 and often find myself going to the taskbar to check on a print to find Orca has crashed. When viewing the log it usually just ends in the middle of something and has nothing related to the crash.

[image]

shayded-exe, 2023-06-09:
> I'm on Windows 11 and often find myself going to the taskbar to check on a print to find Orca has crashed. When viewing the log it usually just ends in the middle of something and has nothing related to the crash.
> 
> [image]

This is exactly what I'm experiencing. Log lines are half cut off after a crash.

julie777, 2023-06-09:
Cutting off the logs is not a big surprise. Most if not all of the crashes that I have seen are in the two BBL DLLs that are downloaded into the plugins directory at run time. On Windows these are dynamically loaded and there is no way that I can find in Windows Dev documentation to be sure to catch all failures in a DLL and prevent the application from crashing. It is sad that BBL is keeping the code proprietary and preventing us from fixing the problems.

limited660, 2023-06-09:
> Cutting off the logs is not a big surprise. Most if not all of the crashes that I have seen are in the two BBL DLLs that are downloaded into the plugins directory at run time. On Windows these are [...]

This is either a recent update issue or Windows 11 issue...not sure but I know I didn't have this problem previously.

IrishMoss, 2023-06-09:
> This is either a recent update issue or Windows 11 issue...not sure but I know I didn't have this problem previously.
I'm still on windows 10, and have the issue.
Originally it was with v1.5 of orca.  The upgrade to 1.6 appeared to fix the issue, and i didn't have crashes for a while, and been running 1.6.2 without issue.
Then it just started happening again in the past week or so.  Upgraded to 1.6.3, but issue remains.
Most recent windows update was on 5/17, so that doesn't appear to be it either.
Really is a mystery... :(

julie777, 2023-06-19:
I have had this problem for months on Windows 10. It was happening just as much on 1.6.2.

closed 2023-11-25 by github-actions[bot]