OrcaSlicer/OrcaSlicer #390: Print sequence: By colour

rkarlsba, 2023-02-27:
Currently, we can choose between print sequence by layer or object. This is fine, but if you have, say, five blue objects and five red ones, using by layer obviously won't make much sense, because it'll drain a lot. Also, by object will move the parts so far from oneanother you can hardly print a few. Now, if you were to group these by colour, that whole set could be printed without issues and then, after that, moving to another colour with the spacing needed. This shouldn't be too hard, but would be brillant for mass production!

SwagDad99, 2023-03-11:
You could put all the objects of the same color near each other, select them all, right-click, and choose assemble. Then Softfever treats them as one object, and then can all huddle together in a corner... Other objects of like color can be grouped, and sent to their own corner...  Lots of objects in each corner now.  Now, it is the same as print by color...

rkarlsba, 2023-03-11:
Sure, it's a possibility, but then I can't rearrange as easily as if the slicer thought of this in the first place. It doesn't look like a big thing to fix, after all.

rkarlsba, 2023-06-18:
However… if it's not much pain, could you please add this feature?

rkarlsba, 2023-11-17:
hm… could someone please look into it, then? I mean - it shouldn't be much work

closed 2024-02-25 by github-actions[bot]