OrcaSlicer/OrcaSlicer #392: Request: Add Ender 5 Pro Support

mssmison, 2023-02-27:
Would you be able to add the Ender 5 pro profile?

bojangles86, 2023-03-01:
> 

or all the creality ones.. if not then just the most used.. E3, E5(all of them,E7.

jeanres, 2023-03-05:
CR 10 Smart would be great as well

mm108, 2023-05-28:
> > 
> 
> or all the creality ones.. if not then just the most used.. E3, E5(all of them,E7.

CR-10 Smart Pro too :)

neclimdul, 2023-11-12:
I think there's technically a Ender 5 Pro profile now so probably can close this.

garocdev, 2023-11-21:
> I think there's technically a Ender 5 Pro profile now so probably can close this.

Tried the one listed as Ender 5 pro 2019 - can't get it to work.  First layer goes down fine, but then something happens and remaining layers are spaghetti or filament. Working with Merlin firmware 2.1.  Using Cura without issue.

closed 2024-02-27 by github-actions[bot]