OrcaSlicer/OrcaSlicer #393: Overhang detection deactivated with active supports.

foxmckraut, 2023-02-28:
**Describe the bug**
The overhang detection isnt working when supports are activatet > results in bad overhangs in unsupportet regions

**Printer model**
x1c

**Desktop (please complete the following information):**
 - Macosx arm BS-SF 1.4.5

[image]
[image]

foxmckraut, 2023-02-28:
Problem solved. if topz distance is set to 0 it disables overhangs

[image]
[image]

SoftFever, 2023-03-02:
Thanks for the report and updates.

closed 2023-11-25 by github-actions[bot]