OrcaSlicer/OrcaSlicer #394: WhamBam Start/End Gcode For Build Plate Compatibility

Ggreiner, 2023-02-28:
I don't know if this has been addressed or not before, but I'm using Softfever with my X1 Carbon and want to purchase the WhamBam build plate.  For the plate to work properly, the reviewers say you have to copy/paste new start/end GCODE into the slicer.

But, I'm concerned about doing this as I don't know whether Softfever has already edited the start/end GCODE and this action will erase your settings.  My question:  Is it possible to get a version of Softfever start/end GCODE that accommodates the WhamBam plate?

WhamBam Install Instructions:
https://img1.wsimg.com/blobby/go/e609d0dd-0e9a-451a-9623-1053f671b776/downloads/Bambu%20X1%20instructions%20v3.pdf?ver=1675136136655

New GCODE:
https://www.printables.com/model/322000-bambu-lab-modified-start-and-end-gcode

Thank you!

PS. I'm a noob to the X1 carbon so I apologize if I haven't even asked the right questions.

barisahmet, 2023-03-01:
I am using WhamBam without any problem with default gcode settings. It does work just perfect.

SoftFever, 2023-03-29:
It is up to user to change it.
The slicer will stay as general as possible.

closed 2023-03-29 by SoftFever