OrcaSlicer/OrcaSlicer #396: Arch Linux launch failure: IOT instruction (core dumped)

coopbri, 2023-02-28:
**Describe the bug**
Upon launching the latest bundled release (AppImage, v1.4.5), a prompt to use a certificate is displayed, and the program crashes after selecting "Yes" with invalid pointer error. Trace:

```
[2023-02-28 15:56:54.508831] [0x00007f85b6ec0f40] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/brian/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so
error, dlerror is /home/brian/.config/BambuStudio-SoftFever/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:25619): Gtk-CRITICAL **: 15:56:58.818: gtk_window_resize: assertion 'width > 0' failed

(bambu-studio:25619): GLib-GObject-WARNING **: 15:56:58.838: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:25619): Gtk-CRITICAL **: 15:56:58.838: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[...]

(bambu-studio:25619): Gtk-CRITICAL **: 15:56:59.102: gtk_widget_set_size_request: assertion 'width >= -1' failed
free(): invalid pointer
[1]    25619 IOT instruction (core dumped)  ./BambuStudioSoftFever.AppImage
```

**3mf File for This Bug**
N/A

**To Reproduce**
- Download [v1.4.5 release](https://github.com/SoftFever/BambuStudio-SoftFever/releases/tag/v1.4.5) (AppImage in Linux zip)
- Execute AppImage
- Select "Yes" in system certificate prompt

**Expected behavior**
SoftFever Bambu Studio launches successfully

**Screenshots**
For some reason, I am unable to upload screenshots to GH at this time (PNG, JPG  -- "Something went really wrong, and we can't process that file") -- will update this issue as soon as I am able to.

**Printer model**
N/A

**Desktop (please complete the following information):**
 - **OS:** Arch Linux, KDE Plasma v5.27.1
 - SoftFever v1.4.5

psi-4ward, 2023-04-16:
Same here. Any ideas how to fix this?

coopbri, 2023-06-30:
@psi-4ward sadly not yet :( Been running Orca Slicer on Windows in meantime

https://github.com/SoftFever/OrcaSlicer/issues/1085 appears to be similar

kzuahelp, 2023-08-15:
Fixed in BambooStudio 1.7 https://github.com/bambulab/BambuStudio/issues/1782#issuecomment-1630290180

Nosnahc, 2023-09-08:
I have the same error too.

llitz, 2023-09-12:
> Fixed in BambooStudio 1.7 [bambulab/BambuStudio#1782 (comment)](https://github.com/bambulab/BambuStudio/issues/1782#issuecomment-1630290180)

That's a very different error from this one, here the app doesn't even open, that error happens after the app opens

closed 2023-12-19 by github-actions[bot]