OrcaSlicer/OrcaSlicer #397: [Bug] Arachne generator and PA tower are not playing nice, creating bad paths for tuning

psiberfunk, 2023-02-28:
**Describe the bug**
As of the latest version, the Arachne wall generator (now default) and the PA tower calibration is producing a GCODE pathing that is not suitable for calibration.  In particular, in the attached 3MF file and screenshot the Arachne generator can be seen to produce lots of little corner fills that screw up the previous 3-line (2x perimeters+gap) method used to successfully calibrate PA values.

Because this introduces retractions and un-retracts and small draws in precisely the area we are focused on tuning, it introduces a lot of possible sources of other error into the PA tuning tower and is very undesirable.

The PA tuning tower routine should either compensate for this in the slicing, the model itself, or should set "classic" wall generation to get good tuning results (not sure what the best fix is here).

Proper (classic) slicing looks like this:
[image]

Bad (arachne) slicing looks like this (yellow corner inner-perimeters drawn first:  
[image]

**3mf File for This Bug**
[Bad_Arachne_Pressure Advance Test.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10855581/Bad_Arachne_Pressure.Advance.Test.zip)

**Desktop (please complete the following information):**
 - OS: Windows 
 - Version 1.4.5  (also nightlies)

psiberfunk, 2023-03-01:
This appears to be the same issue as #353

psiberfunk, 2023-03-02:
Adding photo of (left to right):

Arachne original
Arachne + 25% wall transition threshold angle fix 
Classic.

The latter two are close, but "classic" I still think is best and leads to the most accurate calibration here as it does not include weird gaps on a calibration print which is sensitive to gaps.

I would strongly recommend the PA tower be printed in Classic mode.

[image]

SoftFever, 2023-03-02:
Thank you for the test. 
I will update the Arachne setting for the PA tower. 
However, please note that the PA tower is designed to simulate actual printing with only a few adjustments made to amplify the artifacts
The user's choice of perimeter generator in the profile should be respected.

psiberfunk, 2023-03-02:
> Thank you for the test. I will update the Arachne setting for the PA tower. However, please note that the PA tower is designed to simulate actual printing with only a few adjustments made to amplify the artifacts The user's choice of perimeter generator in the profile should be respected.

Generally I would agree about respecting user choice... however in this case I think makes sense to either warn the user, or, if they're using Arachne , to at least set the 25%.

I will note that you changed the default wall generator to Arachne .. so it's really not the user's default choice.. it's the slicer's that caused this ;)

closed 2023-03-17 by SoftFever