OrcaSlicer/OrcaSlicer #399: Tool head crashes when printing by object and an error occurs

SwagDad99, 2023-03-01:
**Describe the bug**
Printing by object, one object in each of the corners.  After the first two print, back right, and front right, an error happens when printing the back left object.  The toohead crashes into the front right object while trying to resume.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Print by object, with 'recover lost steps' ON in print options.
2. Print by object, an object in each corner of the bed, all the same height.
3. Tool head cover comes off while priting the left rear object.
4. When the cover is replaced, and 'Resume' is clicked, the tool head attempts to 'Home' in the front right of the bed, crashing into the objects already printed.
5. In order to resume printing, the two completed objects must be removed to prevent collisions with them.

**Expected behavior**
the bed should drop at least the amount of the maximum 'Z' height of the highest completed object before attempting to 'Home' the tool head.  THEN raise the bed back up to the previous height, and continue printing.

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [11 Home]

hliebscher, 2023-03-01:
is this in org BambuStudio the same?

SwagDad99, 2023-03-01:
Yes, I posted it in both places.

D

From: Heiko Liebscher 
Sent: Wednesday, March 1, 2023 12:07 AM
To: SoftFever/BambuStudio-SoftFever 
Cc: SwagDad99 ; Author 
Subject: Re: [SoftFever/BambuStudio-SoftFever] Tool head crashes when printing by object and an error occurs (Issue #399)

is this in org BambuStudio the same?

closed 2023-11-26 by github-actions[bot]