OrcaSlicer/OrcaSlicer #400: P1P fans are not working

mike1808, 2023-03-01:
**Describe the bug**
Aux and part cooling fan are not working during printing.

**3mf File for This Bug**
[bug.3mf.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10857851/bug.3mf.zip)

**To Reproduce**
1. Slice any model
2. In the preview make sure cooling is on.
3. Send to the printer.

**Expected behavior**
Fans should follow slicer preview.

**Screenshots**
[image]

**Printer model**
Bambu P1P

**Desktop (please complete the following information):**
 - OS: Linux workstation 5.15.91-1-MANJARO
 - Version: built from https://github.com/SoftFever/BambuStudio-SoftFever/tree/v1.4.5

asoderberg, 2023-03-01:
I used the .3mf file provided by @mike1808 and tested this on my P1P using his steps to reproduce. While the part cooling fan worked as expected, the auxiliary fan never came on during the print (it is set to 70% in the filament profile).

**Printer model**
Bambu P1P 
Firmware: 01.01.01.00

**Desktop:**

- OS: MacOS Ventura 13.2.1 on my 2021 MacBook Pro (M1)
- Version: Downloaded 1.4.5 [BambuStudio-SoftFever_V1.4.5_Mac_arm64.zip](https://github.com/SoftFever/BambuStudio-SoftFever/releases/download/v1.4.5/BambuStudio-SoftFever_V1.4.5_Mac_arm64.zip)

asierralozano, 2023-03-02:
Same issue here. I've used a simple cube created using the `Add primitive` option in the slicer, turned on the Aux fan cooling, and it is always off during the printing

asierralozano, 2023-03-02:
Ok, found the issue. If you go within the machine settings, there is an option named `Aux fan whatever`. Turning that on, fixes the issue

SoftFever, 2023-03-02:
> 

That's right. Aux fan is optional for P1P.
So by default, the P1P Aux fan is disabled until the user manually tick that box in machine settings.

closed 2023-03-05 by SoftFever