OrcaSlicer/OrcaSlicer #402: [Feature request] Brim ears

liftbag, 2023-03-01:
**Is your feature request related to a problem? Please describe.**
There are cases where brim ears are extremely useful, or save time and material.

**Describe the solution you'd like**
If there were no particular compatibility issues, porting from SuperSlicer could be considered, as it seems to me to be very well implemented, with useful configuration parameters.

**Additional context**
Here is an example of use. The material used has warping problems, but strong adhesion between perimeters and layers. Due to warping, I have to keep a distance between brim and object of 0mm. This causes a problematic separation between the brim and the piece. Using SuperSlicer's brim ears allowed me to easily remove the ears from the piece with a cutter.

[image]

[image]

Fredyy90, 2023-03-01:
I only knew about manually placed anti-warping tabs using a cura plugin, but this looks like a really neat solution, if the automatic placement is clever enough

liftbag, 2023-03-01:
> I only knew about manually placed anti-warping tabs using a cura plugin, but this looks like a really neat solution, if the automatic placement is clever enough

The ears in the screeshots are automatically generated by SuperSlicer. I changed the max angle (170°) to increase the number of ears. And I chose the rectilinear pattern, because the concentric produced strange results.

StefanX13, 2023-03-03:
Would be much appreciated, since i use them a lot for thin ABS Parts which tends to warp pretty easy.

guillaumemolter, 2023-03-03:
+1 actually requested samilar feature on [Vanilla Bambu Studio](https://github.com/bambulab/BambuStudio/issues/1393). I would personally prefer a manual paint feature like [TabAntiWarping](https://github.com/5axes/TabAntiWarping) but I would gladly take an automated feature if the implementation is faster/easier.

thepowersource, 2023-03-05:
I would really like to see this feature also as the traditional brim is not really necessary because the prints always warp in certain places not around the whole object.

operinko, 2023-04-28:
I would also love to see this implemented.

H4kunaM4tata, 2023-05-03:
That is a nice feature, I manually add cylinder with the same layer height like 0.2
It is manual and all but works somewhat haha
+1 for this feature

cryoglobe, 2023-05-31:
> +1 actually requested a similar feature on [Vanilla Bambu Studio](https://github.com/bambulab/BambuStudio/issues/1393). I would personally prefer a manual paint feature like [TabAntiWarping](https://github.com/5axes/TabAntiWarping) but I would gladly take an automated feature if the implementation is faster/easier.

Yes .. SuperSlicer has this "Brim ears" and automatically generated for a while , that would be a real added value in Orca 
What could be nice to implement since the beginning , is the possibility to select any "ear" and be able to delete , or reduce or enlarge it individually to customize each if needed ( I'm thinking about some "print in place" prints )

Linnie73, 2023-05-31:
> >  ( I'm thinking about some "print in place" prints )

Any articulated print is dramatic for brim removal, even if you don't have elephant's foot and a proper brim distance. IF the designer knows their stuff, you shouldn't even need brims to get the print to stick to the bed, but I've had varying experiences. 

So +1. specifically for mouse ears to be placed either automatically OR manually painted, and for both to behave according to the normal brim rules, i.e. adjustable width and brim distance, and placement around the object instead of across the bottom (like a flattened cylinder does).

guestisp, 2023-06-19:
+1 for this.
very useful when printing single walls RC planes

Fredyy90, 2023-06-29:
+1 any updates on this request? this would be the killer feature to make orcaslicer the clearly superior fork.

cybrsage, 2023-07-05:
I would love this, as some prints tend to lift on a single corner and brim can be a real hassle to remove.

mlewis-everley, 2023-07-31:
+1, I use this feature in SuperSlicer a lot and it is probably the main reason I can no longer get on with any other slicer :-o

FireflyJar, 2023-08-07:
+1  I could use this too.

codeframe78, 2023-08-23:
+1 I legit just came here to post about this very request LOL.

FireflyJar, 2023-08-23:
> +1 I legit just came here to post about this very request LOL.

it is in the latest release now.   Thank you Noisyfox!

operinko, 2023-08-24:
I believe this FR can be closed, as brim ears have been implemented as an option now.

liftbag, 2023-08-24:
Thanks a lot.

closed 2023-08-24 by liftbag

Knaudler, 2024-03-18:
+1 
need Brim ears

cryoglobe, 2024-03-18:
> +1 need Brim ears

it's already in Orca , this topic is closed

HakunMatat4, 2024-03-18:
> > +1 need Brim ears
> 
> it's already in Orca , this topic is closed

Since when???
AFAIK OrcaSlicer only support it everywhere rather than only where I need.

cryoglobe, 2024-03-19:
> > > +1 need Brim ears
> > 
> > 
> > it's already in Orca , this topic is closed
> 
> Since when??? AFAIK OrcaSlicer only support it everywhere rather than only where I need.

Hello, then you need to open another feature request ...  "paint brim ear"   that would be indeed cool to paint only where you wants brim ears 
or maybe have the choice :  paint where you wants brim ears , or paint where you do not wants them