OrcaSlicer/OrcaSlicer #405: Height Range Modifiers

djngnear, 2023-03-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

It's super helpful to use height range modifiers to have more control over the slice. I'm frustrated when I want to change certain settings (layer height, infill, wall count) for select layers but have to try to line up primitive shapes to modify the objects.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Implementing height range modifier settings where settings for certain ranges or layers can be changed would be amazing. This feature is one of the only reasons I'll slice in prusaslicer/superslicer from time to time.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

An alternative would be lining up modifier shapes and that can be time consuming. 

**Additional context**
Add any other context or screenshots about the feature request here.

pemerick07, 2023-06-22:
This would be a great addition. I really like how Superslicer allows you to change settings in for specific height ranges, particularly layer height. It frustrates me that in Bambu/Orca I can't set a height range with a specific layer height, instead it forces me to fade from one layer height to another in a relatively imprecise way.

djngnear, 2023-06-22:
> This would be a great addition. I really like how Superslicer allows you to change settings in for specific height ranges, particularly layer height. It frustrates me that in Bambu/Orca I can't set [...]

They both have it now.

closed 2023-11-19 by SoftFever