OrcaSlicer/OrcaSlicer #407: FEATURE REQUEST: Add Snapmaker Artisan as 3th party printer

Obicom, 2023-03-02:
**Is your feature request related to a problem? Please describe.**
Please add the Snapmaker Artisan as a 3th party printer

**Describe the solution you'd like**
Snapmaker Artisan should be an printer option in Bambu Studio SF

**Describe alternatives you've considered**
No alternative

**Additional context**
[image]

hliebscher, 2023-03-02:
you can use the snapmaker 350 profile an rename it and set the right dimensions ... for dualprint head you need only add a second color.
works with Artisan and the J1 very easy

troian, 2023-04-18:
I think there is a bit more than just renaming and dimensions (tho certainly it will work).
- the print area is different depending on which nozzle is used
- L & R nozzles can have different diameter

ZeroMuhi, 2023-09-11:
will a Snapmaker Artisan printer be added for Orca Slicer?

closed 2023-12-18 by github-actions[bot]