OrcaSlicer/OrcaSlicer #409: Arc overhang support or compatibility with arc overhang script

azhago, 2023-03-02:
Arc overhang is a new way to generate overhang, pretty impressive.
https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration

Script works with prusaslicer, but not with bambulab studio.
Some tags may be changed in the Gcode ("Bridge infill" instead of "Bridge" ..)

hliebscher, 2023-03-02:
@SoftFever @azhago  i have changed the python script now working for Bambu-Softfever
[prusa_slicer_post_processing_script.py.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10874335/prusa_slicer_post_processing_script.py.zip)

azhago, 2023-03-02:
Just tried, unfortunately, original script work with puraslicer, but modified script do not with bambulab (same object) 
[testoverhang_object.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10875747/testoverhang_object.zip)

hliebscher, 2023-03-02:
i need your prusaslicer settings  start end Gcode layer change profile settings...the best ist the prusaslicer g-code before the script is running as zip

azhago, 2023-03-04:
@hliebscher 
Here it is, before and after script.
[testoverhang_object_before_script.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10888049/testoverhang_object_before_script.zip)

hliebscher, 2023-03-05:
i have to modify bambustudio ... on time this week sorry

SoftFever, 2023-03-13:
@hliebscher, did you manage to look at what needs to be changed to support the script by any chance?
I should be able to help looking into this issue tomorrow.

astrof, 2023-03-21:
A little late to the party but I wonder if one could look at [https://github.com/pleccer/Pleccer](url) for this; Pleccer is also based off of PrusaSlicer (a true fork as opposed to BambuStudio).

kostecky, 2023-09-21:
Has anyone got the script working with orca? What is the latest status? Thanks!

closed 2023-11-19 by SoftFever

GuzziRaz, 2023-11-29:
@SoftFever is this done or is it not planned? Please comment! Pardon if I missed some info elsewhere

ImBehinYa, 2023-12-23:
How is it looking?

starschild86, 2024-01-22:
Would also like to know 😁

thepkc, 2024-02-06:
This feature would seriously help with some more complex overhangs that almost systematically fail or cause low quality prints. I don't see any code commit as of today that this is done so I would guess this isn't done?

pagdot, 2024-02-13:
Found https://github.com/Kelsch/arc-overhang-orcaslicer-integration
Haven't tried it as I'm still waiting for my 3d printer, but want to give it a try at least. It seems like since the fork a few additional commits were made to the prusaSlicer script. Would be interesting to apply them here too

TheOverpassArsonist, 2024-11-15:
Has there been any reason give why this was closed? It was seemingly just silently marked as not-planned and closed without any real reason, (at least none that I can find) which is particularly odd since it fixes a pretty major problem with 3d printing as a whole fairly elegantly and it's already had several successful practical implementations to reference. (though as mentioned in #4621 existing implementations still definitely can be improved upon) Small or niche features simply not being worth the effort to implement is one thing, but this seems like a pretty major feature that just got ghosted for seemingly no reason.