OrcaSlicer/OrcaSlicer #411: printable area

Anothermbdusted, 2023-03-03:
**Describe the bug**
when going to your printers bed size in dark mode the pop up for the bed size is not in dark mode but a mix of dark and light modes

**To Reproduce**
i setup my own printer none bambu and  when you select the bed size in dark mode its always there

**Expected behavior**
should be in dark mode not a mix of normal and dark

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
home built

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows] 10 home 64 bit
 - Version [e.g. 22] 22h2
    19045.2673    build

hliebscher, 2023-03-03:
Which softfever software version are you referring to, and which printer specifically is this issue occurring on? Also, it would be helpful if you could provide a screenshot.

DVSVIDEO, 2023-03-09:
This is what he meant [Anothermbdusted]
[image]

closed 2023-11-26 by github-actions[bot]