OrcaSlicer/OrcaSlicer #413: Printing of slopes is missing the Bidge Layer

neorm, 2023-03-03:
**Describe the bug**
If parts are printed with a slope in some cases the top surface gets messy. The Slicer is printing Internal solid Infill on the Infill layer, due to the Speed of the Internal solid Infill it's not working well. It should start with a Bridge Layer first, in some cases, the slicer is doing it.
I have made a video to illustrate the problem
https://youtu.be/SWV-gBCttPM

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
[Problem slope.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10880245/Problem.slope.zip)

**To Reproduce**
Slice attached Part.

**Expected behavior**
Bridge Lines should be printed first on the Infill, and on top of that comes the Internal solid Infill.

**Screenshots**
[image]

**Printer model**
see ZIP File

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Bambu Studio SoftFever 1.4.5 / Bambu Studio 1.4.2.13

SoftFever, 2023-03-03:
ya, it's bug

igiannakas, 2023-11-02:
This is now fixed in the latest 1.8 release following the ports from Noisyfox.

Jo-Con-El, 2023-12-05:
@neorm: Could you please verify it and close the issue?  Thanks!

igiannakas, 2023-12-27:
This is not fixed in 1.9 or earlier. It's a bug with the vertical shell implementation. It should first trigger a bridge layer and then run solid infill above it. Present also in the latest prusa slicer. My hunch is that the process_external_surfaces() function just doesnt recognise this as a bridge to expand for some reason.

igiannakas, 2023-12-27:
@SoftFever 
[image]

Some progress made on this.

In the print object.cpp -> bridge_over_infill function there is some filtering going on to remove areas where internal bridging may be unnecessary as the infill is thought to be supported adequately from the sparse infill below.

Commenting these lines below out always does internal bridging irrespective of size of the internal overhangs.

```
                lower_layer_solids = shrink(lower_layer_solids, 1 * spacing); // first remove thin regions that will not support anything
                lower_layer_solids = expand(lower_layer_solids, (1 + 3) * spacing); // then expand back (opening), and further for parts supported by internal solids
                // By shrinking the unsupported area, we avoid making bridges from narrow ensuring region along perimeters.
                unsupported_area   = shrink(unsupported_area, 3 * spacing);
                unsupported_area   = diff(unsupported_area, lower_layer_solids);
```

igiannakas, 2023-12-27:
Funnily enough all slic3r derivative slicers do this as I bet the same filtering code is re-used across Prusa, Bambu and SuperSlicer! Orca may be the first one to "fix" this :D

igiannakas, 2023-12-27:
I'll add an option to "Don't filter small internal bridges" and push a PR.

neorm, 2023-12-28:
Hi, I'm happy to see that someone is working on it. I'm not at home at the moment. But I'll be back at the weekend, if there is anything i should check let me know. Last time I was checking it the bug was still in there. (2-3 Weeks ago)

igiannakas, 2023-12-28:
> Hi, I'm happy to see that someone is working on it. I'm not at home at the moment. But I'll be back at the weekend, if there is anything i should check let me know. Last time I was checking it the bug was still in there. (2-3 Weeks ago)
> 
> edit: Ok, I found the build, will test it when I'm home. I'm so hyped about this, thank you for taking care of it!

It should be fixed in this build.

however depending on the infill selected and the density you will get bridging but it may leave a 1 layer gap between it and the solid infill. That shouldn’t matter much at this point as worst case it is supported slightly lower. And much better than before.

igiannakas, 2023-12-28:
If you do any tests please post them in the above PR.

neorm, 2023-12-28:
> If you do any tests please post them in the above PR.

ok, i will

closed 2024-01-13 by SoftFever