OrcaSlicer/OrcaSlicer #415: Softfever not slicing correctly - adding extra lines to prints

parrothd, 2023-03-03:
**Describe the bug**
When using softfever to slice a 3d print with embossed texted the embossing or texted is filled incorrectly when prints. 

When I slice with Bambu the text is sliced and prints correctly. 

Both slicers are using the same default settings.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
open and slice file, see the attached images, text is filled where it should not be.

**Expected behavior**
The print should look like the model, embossed text.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
p1p
**Desktop (please complete the following information):**
windows 10, latest patches, newest softfever 1.4.5

SoftFever, 2023-03-04:
Hi @parrothd ,

Default settings are slightly different between SF and BambuStudio, you can try switching to the classic wall generator instead of Arachne. That might solve your problem.

parrothd, 2023-03-04:
[image]

I changed softfever to classis and still has the same issue, see screen shot.

In Bambu I switched to Arachne and it still sliced correctly.

hliebscher, 2023-03-05:
can you attach the 3mf project file from bs-sf as zip?

parrothd, 2023-03-06:
It's the auto brim feature in softfever, if I change it to none it works all other options have the same issue. 

Strange that it's only on some text, and see the attached image, I see these tiny brims affects when using softfever with auto. brim. Could be something from fusion?

[image]
 

[Softfever-Text-Issue.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10903017/Softfever-Text-Issue.zip)

SoftFever, 2023-03-12:
> It's the auto brim feature in softfever, if I change it to none it works all other options have the same issue.
>
> Strange that it's only on some text, and see the attached image, I see these tiny [...]

Thanks for the update.
Auto brim doesn't work well in many cases.

closed 2023-11-26 by github-actions[bot]