OrcaSlicer/OrcaSlicer #420: Missing layers when slicing with latest BambuStudio

Guilouz, 2023-03-03:
**Describe the bug**

Missing layers when slicing.

**File for This Bug**

Tested with many files in different format (stl, 3mf, step), same issue.

[35x35_0.4mm_nozzle_flowrate_calibration.stl.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10886637/35x35_0.4mm_nozzle_flowrate_calibration.stl.zip)

**To Reproduce**

Just install latest BambuStudio on macOS

**Screenshots**

[image]

[image]

No issue with SofFever fork :

[image]

**Printer model**

Bambulab X1 Carbon

**Desktop**

 - OS: MacOS Ventura

SoftFever, 2023-03-04:
Hi @Guilouz ,

Did you mean it's a bug in BambuStudio only?
If that's the case, I'd suggest you post this on BambuStudio's repo.

Guilouz, 2023-03-04:
Ahh yes sorry! I have posted this in wrong repo. Sorry.

closed 2023-03-04 by Guilouz