OrcaSlicer/OrcaSlicer #424: Not opening in MacOS 13.2.1 - xattr error

thepowersource, 2023-03-04:
**Describe the bug**
App will not open despite providing the correct xattr command in terminal. Also tried "override malware protection" from get info menu but still get the same error "The application BambuStudio-SoftFever can't be opened" and "BambuStudio-Softfever will damage your computer. You should move it to the bin".

Downloaded app installed fine then 1 week later it stopped working and gave an error of malware. No changes to MacOS or any malware tools it just simply stopped working. Tried to delete and reinstall but same error. Tried terminal xattr -d com.apple.quarantine/Applications/BambuStudio-SoftFever.app but now get the following error "Not enough arguments for option -d. Expected at least 2 but got 1"

Carbon X1
MacOS 13.2.1

hliebscher, 2023-03-05:
Have you read the note on the Main page about this issue and the articel that are linked? https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/

same at https://github.com/SoftFever/BambuStudio-SoftFever/issues/274

xattr -d com.apple.quarantine /Applications/BambuStudio-SoftFever.app/

thepowersource, 2023-03-05:
> Have you read the note on the Main page about this issue and the articel that are linked? https://www.howtogeek.com/803598/app-is-damaged-and-cant-be-opened/
> 
> same at #274
> 
> xattr -d com.apple.quarantine /Applications/BambuStudio-SoftFever.app/

Downloaded app installed fine then 1 week later it stopped working and gave an error of malware. No changes to MacOS or any malware tools it just simply stopped working. Tried to delete and reinstall but same error. Tried terminal xattr -d com.apple.quarantine/Applications/BambuStudio-SoftFever.app but now get the following error "Not enough arguments for option -d. Expected at least 2 but got 1"

hliebscher, 2023-03-05:
copy pls and insert in terminal
xattr -dr com.apple.quarantine /Applications/BambuStudio-SoftFever.app

thepowersource, 2023-03-05:
> copy pls and insert in terminal
> xattr -dr com.apple.quarantine /Applications/BambuStudio-SoftFever.app

Copied and pasted updated command into terminal. Looked like it worked, no error from xattr but app still will not open and gives same macOS warning. What is strange is that it was working fine just 2 days ago. I have not changed anything or had any updates so I am confused as to why it just stopped working.

GP3D84, 2023-03-05:
Did you try the previous command with sudo before?

thepowersource, 2023-03-05:
Yes

> Did you try the previous command with sudo before?

Yes and it still has the same problem. What I can't figure out is why it just stopped working. Last time I just entered xattr -d com.apple.quarantine /pathname, it was working fine then just stopped... Now I keep getting "The application BambuStudio-SoftFever can't be opened" and "BambuStudio-Softfever will damage your computer. You should move it to the bin"

thepowersource, 2023-03-05:
When I click get info and override the malware protection it says "Are you sure you want to allow opening “BambuStudio-SoftFever”? It contains malware." This warning is different to the normal "The application BambuStudio-SoftFever can't be opened" and "BambuStudio-Softfever will damage your computer. You should move it to the bin". I ignore the warning but still will not load the app.

mmackowiak, 2023-03-05:
Same issue here, no combination of params/sudo usage for `xattr` is helping.

thepowersource, 2023-03-06:
> Same issue here, no combination of params/sudo usage for `xattr` is helping.

I also tried various xattr params and nothing. It's really odd because it worked before and I didn't change anything. 

I would pay for a working version in the App Store.

ab3dgeek, 2023-03-06:
same issue came up just out of the blue yesterday for me too.
(there might have been some security updates to the os, I'm not sure)

I solved it by deleting the arm version, copying the x86_64 in its place and I then run the xattr command in terminal and it works again.

thepowersource, 2023-03-06:
> same issue came up just out of the blue yesterday for me too. (there might have been some security updates to the os, I'm not sure)
> 
> I solved it by deleting the arm version, copying the x86_64 in its place and I then run the xattr command in terminal and it works again.

Well done ab3dgeek it is now working again. Super slow though. I wonder when we can have the dedicated ARM version working? Next update maybe? Could someone pass the message to the developer? He should just release a paid version, I would pay $25 for the added functionality over the standard slicer.

olsonjj, 2023-03-06:
Same issue here. This was working yesterday, and now its not.  Trying to re-run the code fails with error "xattr: /Applications/BambuStudio-SoftFever.app/: No such xattr: com.apple.quarantine"   Pretty sure this has something to do with the recent security update.

esc247, 2023-03-06:
Check this issue out: https://github.com/SoftFever/BambuStudio-SoftFever/issues/427

It works if you use the non arm version. Also I was facing the same issue, and a good ol restart did the trick lol.

olsonjj, 2023-03-06:
> Check this issue out: #427
> 
> It works if you use the non arm version. Also I was facing the same issue, and a good ol restart did the trick lol.

Running a slicer in rosetta is a no-go for me. Its already slow on bigger models as it is. I'll switch back to Bambu Studio until this is resolved. 

I'd be happy to help part of the $99/yr Apple dev license.

ScourFly, 2023-03-07:
Version 1.4.4 runs without problems then updated to 1.4.5 and nothing works anymore. Delete everything and installed again 1.4.4 with the description (xattr -dr com.apple.quarantine /Applications/BambuStudio-SoftFever.app) and everything works again!

ab3dgeek, 2023-03-07:
> Well done ab3dgeek it is now working again. Super slow though. I wonder when we can have the dedicated ARM version working? Next update maybe? Could someone pass the message to the developer? He [...]

Thanks, I did notice it took longer to load in but otherwise I only used it on the laptop for monitoring and didn't notice any performance issue doing that. ( I sliced on my 2015 intel iMac which had no problem with softfever at all)

ab3dgeek, 2023-03-07:
> > > > same issue came up just out of the blue yesterday for me too. (there might have been some security updates to the os, I'm not sure)
> > > > I solved it by deleting the arm version, copying the x86_64 in its place and I then run the xattr command in terminal and it works again.
> > > 
> > > 
> > > Well done ab3dgeek it is now working again. Super slow though. I wonder when we can have the dedicated ARM version working? Next update maybe? Could someone pass the message to the developer? He should just release a paid version, I would pay $25 for the added functionality over the standard slicer.
> > 
> > 
> > Thanks, I did notice it took longer to load in but otherwise I only used it on the laptop for monitoring and didn't notice any performance issue doing that. ( I sliced on my 2015 intel iMac which had no problem with softfever at all)
> 
> You are right it hasn't made much difference to the slice time (not that I can notice anyway) but I certainty thought the load time had increased. It's rare to see the spinning ball these days...

my guess is that it takes longer to load because it needs to be 'translated' by rosetta which handles intel based apps for the new arm processors. this is also why it still works though, it handles some security stuff differently (yay)
but to be honest I can't wait for softfever to be properly signed.

hubertron, 2023-03-15:
Same issue as others. Unable to get application to load. Even following steps above. Will not use rosetta. Seems like a legit issue to be addressed for us Mac users.

SoftFever, 2023-03-15:
> 

Please ref this comment. 
You can download the `nosign` version 
https://github.com/SoftFever/OrcaSlicer/issues/427#issuecomment-1470388041

SoftFever, 2023-03-15:
Closing this ticket, let's consolidate the discussions in #427

closed 2023-03-15 by SoftFever